VoltSim/VoltSim #1: Do I set the Oscilloscope Time?

Awen2345, 2021-12-05:
Oscilloscope must be equipped with a timer Please update the application, right?

Awen2345, 2021-12-05:
Uploading Record_2021-12-05-20-13-45.mp4…

Awen2345, 2021-12-05:
[image]

Awen2345, 2021-12-05:
> [image]

Like Everycircuit

voltsimapp, 2021-12-05:
Hi @Awen2345 , 
Yes changing oscilloscope time is in pipeline. In Everycircuit changing oscilloscope time will also change the timestep of the circuit. its is easier to implement but felt bit odd. so we are looking into an alternative implementation in which changing timescale wont affect the simulation.

voltsimapp, 2021-12-16:
Update on this: In the latest version 0.1.3, you can adjust the timestep by horizontal scrolling on the scope. It not fully polished yet, as it dosent update already exisintg data and dosent show current scale. But its something ;)

Awen2345, 2021-12-16:
Oh Thanks (•‿•)

voltsimapp, 2022-06-29:
This feature is now complete and is available in the Instruments addon. You can adjust both the time-step and can zoom into the scope with scroll.