VoltSim/VoltSim #4: Fix the app next update

Awen2345, 2022-03-03:
1. Save Settings with projects
2. IC 555 missing after project opened
3. Centre Circuits after open projects
This is Bug in apk....

Awen2345, 2022-03-03:
Uploading Record_2022-03-03-22-20-36.mp4…

IC 555 is missing after saving project and open projects

Awen2345, 2022-03-03:
Uploading Record_2022-03-03-22-20-36.mp4…
IC 555 is missing after open saved projects

voltsimapp, 2022-03-03:
Hi, Thanks for reporting the bug. 

1. can you explain this please. Do you mean that the setting like "show voltage", "grid setting" etc should be saved for each project?

2. Are you facing this for all the projects with an IC?

3.The circuit should automatically center once project is loaded. Are you seeing this issue for all the projects?

voltsimapp, 2022-03-07:
Hello @Awen2345 , can you please send a screenshot of the circuit where the IC 555 is missing after saving and opening project, please?

Awen2345, 2022-03-14:
[image]

Before

Awen2345, 2022-03-14:
[image]

Awen2345, 2022-03-14:
[image]
After I reopened My projects

nithinmurali, 2022-03-15:
thanks @Awen2345 , We will fix this in the next update.

voltsimapp, 2022-04-01:
We have a new update released which fixed all these issues reported.

voltsimapp, 2022-04-13:
if the issue persists, please reopen

closed 2022-04-13 by voltsimapp