VoltSim/VoltSim #12: Missing addon

evUser1987, 2022-04-14:
I bought addon with dc motor but I can not use it. Pleas refound money.

evUser1987, 2022-04-14:
[image]

voltsimapp, 2022-04-14:
Hi @evUser1987, 
Looks like a bug, the motor shouldn't be rotated that way.  what extension did you buy? Was the motor rotated like that when you added it? can you please provide some steps to reproduce the issue?

If you wish to have a refund, please email voltsim.app@gmail.com with the extension you bought and the date when it was purchased. We will initiate the refund procedure.

evUser1987, 2022-04-14:
Yes, it was rotated. Chose motor from menu and add... there are bugs in app in projects list also... My order GPA.3369-7900-9211-56900

evUser1987, 2022-04-14:
[image]

voltsimapp, 2022-04-14:
Can you please provide the device model and os information? Are you running in a simulator?

evUser1987, 2022-04-14:
Samsung Galaxy J7 Android 7

evUser1987, 2022-04-15:
Do you have solution??

voltsimapp, 2022-04-16:
We are unable to reproduce this specific issue on our end. So we cannot propose a solution yet.

As mentioned earlier, please email voltsim.app@gmail.com for a refund. Github issues are used only for bugs/feature tracking.

voltsimapp, 2022-06-08:
closing as we couldnt reproduce the bug.

closed 2022-06-08 by voltsimapp