VoltSim/VoltSim #21: Custom IC

UltimateSheep, 2022-11-11:
Hi, voltsim developers

I would like to suggest a feature for adding custom ic. Make a separate tabs for making ic. It is like the normal editor but in custom limited size and the users can add pins to every side. The users can add resistor, leds or logic gates. In normal editor, the users can add the part they have made like normal ic digital parts. You can include it in an extension or make it free.

Happy coding my guys,
 UltimateSheep

voltsimapp, 2022-11-11:
Hi @UltimateSheep, Thank you for the detailed suggestion. A custom components editor like you mentioned is indeed in our pipeline. 

But if we add this we would like to make our simulation engine more efficient. Else it won't be able to simulate large custom devices. Please stay tuned for our future updates. We will update any progress on this in this thread.

voltsimapp, 2022-11-27:
We have added other project imports in 0.1.19. This is a pre-requisite feature to Custom ICs.

UltimateSheep, 2022-11-28:
Hi voltsim developers,

You did a very good job on the simulator and I like it very much. Unlike other apps, the editor is easy to use, extensions are not monthly payment and they are pretty good deal so keep it up developers

Good luck,
UltimateSheep

voltsimapp, 2025-02-17:
Hi, This feature is being tested in closed beta. This will be released soon. Please email voltsim.app@gmail.com if you would like to join the testing and provide early feedback on this.