VoltSim/VoltSim #22: Insecure app content

damiandaminczyk, 2022-11-17:
I found out that this app is insecure, database content can be easily cracked:

`{"max_time_step":5.0E-6,"min_time_step":5.0E-11,"iteration_count":3.9121283998153213,"current_speed":55,"voltage_range":5,"elements":[{"name":"Wire","x":325,"y":500,"x2":325,"y2":550,"height":0,"label":"nullnull","flags":0,"uid":"4a7d6acf-bb31-42fd-bb8d-e21c952ae0aa","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":24,\"width\":101,\"x\":325,\"y\":488},\"second\":{\"height\":4,\"width\":4,\"x\":323,\"y\":548}}","segmentPoints":"[]"},{"name":"Wire","x":325,"y":600,"x2":325,"y2":550,"height":0,"label":"nullnull","flags":0,"uid":"fb519de0-7e1a-4a82-8251-0de40a5b360f","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":32,\"width\":101,\"x\":325,\"y\":584},\"second\":{\"height\":4,\"width\":4,\"x\":323,\"y\":548}}","segmentPoints":"[]"},{"name":"Wire","x":425,"y":500,"x2":425,"y2":550,"height":0,"label":"nullnull","flags":0,"uid":"fdd3b11f-315a-4476-99bf-dc675157cdd8","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":24,\"width\":101,\"x\":325,\"y\":488},\"second\":{\"height\":4,\"width\":4,\"x\":423,\"y\":548}}","segmentPoints":"[]"},{"name":"Wire","x":225,"y":550,"x2":325,"y2":550,"height":0,"label":"nullnull","flags":0,"uid":"47329cbd-ae67-479c-9f26-eeb633a15578","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":101,\"width\":24,\"x\":213,\"y\":450},\"second\":{\"height\":4,\"width\":4,\"x\":323,\"y\":548}}","segmentPoints":"[]"},{"name":"Wire","x":75,"y":600,"x2":225,"y2":650,"height":0,"label":"nullnull","flags":0,"uid":"e932feb1-ee90-48d3-a68a-b07d2a903d6e","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":101,\"width\":34,\"x\":58,\"y\":500},\"second\":{\"height\":101,\"width\":24,\"x\":213,\"y\":550}}","segmentPoints":"[{\"x\":75,\"y\":650}]"},{"name":"Wire","x":75,"y":500,"x2":225,"y2":450,"height":0,"label":"nullnull","flags":0,"uid":"afddee97-40a1-454b-94f1-517d6d585351","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":101,\"width\":34,\"x\":58,\"y\":500},\"second\":{\"height\":101,\"width\":24,\"x\":213,\"y\":450}}","segmentPoints":"[{\"x\":75,\"y\":450}]"},{"name":"Wire","x":425,"y":550,"x2":500,"y2":550,"height":0,"label":"nullnull","flags":0,"uid":"d091b2e2-37a7-4183-ac4a-071da053731c","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":4,\"width\":4,\"x\":423,\"y\":548},\"second\":{\"height\":101,\"width\":24,\"x\":488,\"y\":550}}","segmentPoints":"[]"},{"name":"Wire","x":225,"y":450,"x2":500,"y2":450,"height":0,"label":"nullnull","flags":0,"uid":"5f74b49a-ffcf-44e8-a3bc-8645de0cdd98","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":101,\"width\":24,\"x\":213,\"y\":450},\"second\":{\"height\":101,\"width\":24,\"x\":488,\"y\":450}}","segmentPoints":"[]"},{"name":"Wire","x":425,"y":600,"x2":425,"y2":550,"height":0,"label":"nullnull","flags":0,"uid":"82f5a1bd-a0a6-42f6-9076-9c2c722f94b3","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":32,\"width\":101,\"x\":325,\"y\":584},\"second\":{\"height\":4,\"width\":4,\"x\":423,\"y\":548}}","segmentPoints":"[]"},{"name":"ACVoltage","x":75,"y":600,"x2":75,"y2":500,"height":0,"label":"V3","flags":0,"uid":"c3c04a3e-bf6c-497b-99e2-972e70378381","scoped_values":[0],"waveform":1,"frequency":40,"maxVoltage":5,"bias":0,"phaseShift":0,"dutyCycle":0.5},{"name":"Wire","x":225,"y":650,"x2":500,"y2":650,"height":0,"label":"nullnull","flags":0,"uid":"c0c33bee-5eec-43f2-86cf-b81406fee006","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":101,\"width\":24,\"x\":213,\"y\":550},\"second\":{\"height\":101,\"width\":24,\"x\":488,\"y\":550}}","segmentPoints":"[]"},{"name":"Diode","x":225,"y":550,"x2":225,"y2":650,"height":0,"label":"D5","flags":0,"uid":"b8370952-a52b-4a90-85bd-69c5cc542e5e","scoped_values":[]},{"name":"Diode","x":225,"y":550,"x2":225,"y2":450,"height":0,"label":"D5","flags":0,"uid":"fb4b1f69-2ae5-4bad-98fd-84fe9c0f8311","scoped_values":[]},{"name":"Capacitor","x":325,"y":500,"x2":425,"y2":500,"height":0,"label":"C3","flags":0,"uid":"f51891f2-3564-4c64-8405-f65b0e7944cb","scoped_values":[],"capacitance":1.02E-4,"initialVoltage":0.001},{"name":"Resistor","x":325,"y":600,"x2":425,"y2":600,"height":0,"label":"R3","flags":0,"uid":"2ad68d11-63c6-49c7-b89f-156954124567","scoped_values":[0],"resistance":500},{"name":"Post","x":425,"y":550,"x2":425,"y2":550,"height":0,"label":"nullnull","flags":0,"uid":"dfb7f391-be10-49f8-a747-551ceee88eed","scoped_values":[]},{"name":"Post","x":325,"y":550,"x2":325,"y2":550,"height":0,"label":"nullnull","flags":0,"uid":"e829d08f-c0d7-4703-aa55-8d0e11ad0e9b","scoped_values":[]},{"name":"Diode","x":500,"y":650,"x2":500,"y2":550,"height":0,"label":"D5","flags":0,"uid":"bfc549b1-e518-4095-9eb4-daf345d226e3","scoped_values":[]},{"name":"Diode","x":500,"y":450,"x2":500,"y2":550,"height":0,"label":"D5","flags":0,"uid":"b48ad3db-868d-4f47-8a66-1abaf3b6c375","scoped_values":[]}],"label_map":"{}"}`

`{"max_time_step":5.0E-6,"min_time_step":5.0E-11,"iteration_count":3.9121283998153213,"current_speed":55,"voltage_range":5,"elements":[{"name":"Wire","x":100,"y":550,"x2":125,"y2":525,"height":0,"label":"nullnull","flags":0,"uid":"bd28e780-2869-4af8-824d-2aaa7a02debe","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":101,\"width\":34,\"x\":83,\"y\":550},\"second\":{\"height\":24,\"width\":101,\"x\":125,\"y\":513}}","segmentPoints":"[{\"x\":100,\"y\":525}]"},{"name":"Wire","x":100,"y":650,"x2":125,"y2":675,"height":0,"label":"nullnull","flags":0,"uid":"8f7d64f0-35a9-404d-9f15-0fb023a56c4b","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":101,\"width\":34,\"x\":83,\"y\":550},\"second\":{\"height\":32,\"width\":101,\"x\":125,\"y\":659}}","segmentPoints":"[{\"x\":100,\"y\":675}]"},{"name":"Wire","x":450,"y":550,"x2":450,"y2":750,"height":0,"label":"nullnull","flags":0,"uid":"7c4d39cc-ee53-452e-80cb-de8de598866b","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":24,\"width\":101,\"x\":450,\"y\":538},\"second\":{\"height\":24,\"width\":101,\"x\":450,\"y\":738}}","segmentPoints":"[]"},{"name":"Resistor","x":550,"y":550,"x2":650,"y2":550,"height":0,"label":"R1","flags":0,"uid":"3bd4f708-073b-4ff9-a3cc-2c7afafecffb","scoped_values":[],"resistance":100},{"name":"Diode","x":450,"y":550,"x2":550,"y2":550,"height":0,"label":"D1","flags":0,"uid":"2b8f85dd-5f22-4001-9562-312f57d53a0b","scoped_values":[]},{"name":"Resistor","x":550,"y":750,"x2":650,"y2":750,"height":0,"label":"R1","flags":0,"uid":"4e802f93-f527-4866-9e91-57797470783e","scoped_values":[],"resistance":100},{"name":"Diode","x":550,"y":750,"x2":450,"y2":750,"height":0,"label":"D1","flags":0,"uid":"a2419381-cba1-4f4b-b6c3-af99ee64f95f","scoped_values":[]},{"name":"Voltmeter","x":750,"y":650,"x2":750,"y2":550,"height":0,"label":"nullnull","flags":1,"uid":"81afbb68-48b6-4e54-a6f3-52f8b1480691","scoped_values":[],"value_type":0},{"name":"Voltmeter","x":750,"y":750,"x2":750,"y2":650,"height":0,"label":"nullnull","flags":1,"uid":"32821666-8dc8-41d8-9602-dcc36d4bcaf6","scoped_values":[],"value_type":0},{"name":"Wire","x":650,"y":750,"x2":750,"y2":750,"height":0,"label":"nullnull","flags":0,"uid":"4e3339c7-3cc5-46ca-aafb-68121deb4148","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":101,\"width\":24,\"x\":638,\"y\":650},\"second\":{\"height\":101,\"width\":34,\"x\":733,\"y\":650}}","segmentPoints":"[]"},{"name":"PolarCapacitor","x":650,"y":550,"x2":650,"y2":650,"height":0,"label":"C1","flags":0,"uid":"63f9672f-2960-4ab9-87a1-bf17f69142bc","scoped_values":[],"capacitance":9.999999999999999E-5,"initialVoltage":0.001,"maxNegativeVoltage":1},{"name":"Wire","x":650,"y":550,"x2":750,"y2":550,"height":0,"label":"nullnull","flags":0,"uid":"d5406463-655f-492e-ac1d-65076c463a19","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":101,\"width\":24,\"x\":638,\"y\":550},\"second\":{\"height\":101,\"width\":34,\"x\":733,\"y\":550}}","segmentPoints":"[]"},{"name":"PolarCapacitor","x":650,"y":650,"x2":650,"y2":750,"height":0,"label":"C1","flags":0,"uid":"ee569e59-b4b9-48dd-bec3-a3f186a3a7a3","scoped_values":[],"capacitance":9.999999999999999E-5,"initialVoltage":0.001,"maxNegativeVoltage":1},{"name":"Wire","x":650,"y":650,"x2":750,"y2":650,"height":0,"label":"nullnull","flags":0,"uid":"8598561d-114e-4542-9d13-f76e313f41a5","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":101,\"width\":24,\"x\":638,\"y\":650},\"second\":{\"height\":101,\"width\":34,\"x\":733,\"y\":550}}","segmentPoints":"[]"},{"name":"Wire","x":225,"y":525 [... cut ...] "uid":"6aa91e11-f18c-4873-a336-c677f8fdee44","scoped_values":[],"position":0,"momentary":false},{"name":"Fuse","x":125,"y":525,"x2":225,"y2":525,"height":0,"label":"F1","flags":0,"uid":"c6704fce-909c-4738-bcd0-9adea93846f3","scoped_values":[],"i2t":6.73,"resistance":0.0613},{"name":"Transformer","x":250,"y":550,"x2":400,"y2":550,"height":0,"label":"T1","flags":0,"uid":"749d9195-7f60-4bbc-aef1-0e35d6aed8e5","scoped_values":[],"inductance":4,"turns_ratio":0.02,"coeff":0.999}],"label_map":"{T=1}"}`

voltsimapp, 2022-11-19:
Hi @damiandaminczyk , Thank you for reporting this. In a normal phone, the app database should be inaccessible to the user. Did you use a rooted phone to access the database?

damiandaminczyk, 2022-11-21:
I have acces to database from APK, everything is in plain text. It should be encrypted.

damiandaminczyk, 2022-11-21:
You can check below output from database:
`{"max_time_step":5.0E-6,"min_time_step":5.0E-11,"iteration_count":3.9121283998153213,"current_speed":55,"voltage_range":5,"elements":[{"name":"Wire","x":325,"y":600,"x2":325,"y2":700,"height":0,"label":"nullnull","flags":0,"uid":"97bc4eb0-28d2-4556-91f0-45ee0c14e51d","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":101,\"width\":32,\"x\":309,\"y\":500},\"second\":{\"height\":32,\"width\":101,\"x\":225,\"y\":684}}","segmentPoints":"[]"},{"name":"Wire","x":250,"y":350,"x2":325,"y2":500,"height":0,"label":"nullnull","flags":0,"uid":"0381eeff-00e8-4d69-9bd7-eea4949f48fb","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":32,\"width\":101,\"x\":150,\"y\":334},\"second\":{\"height\":101,\"width\":32,\"x\":309,\"y\":500}}","segmentPoints":"[{\"x\":325,\"y\":350}]"},{"name":"Wire","x":125,"y":350,"x2":150,"y2":350,"height":0,"label":"nullnull","flags":0,"uid":"122e1df6-a6ec-4ec1-972b-6d9fa2affefe","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":4,\"width\":4,\"x\":123,\"y\":348},\"second\":{\"height\":32,\"width\":101,\"x\":150,\"y\":334}}","segmentPoints":"[]"},{"name":"Wire","x":125,"y":350,"x2":125,"y2":475,"height":0,"label":"nullnull","flags":0,"uid":"86b1248a-7089-462a-ad64-05de7c316ce8","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":4,\"width\":4,\"x\":123,\"y\":348},\"second\":{\"height\":101,\"width\":16,\"x\":117,\"y\":475}}","segmentPoints":"[]"},{"name":"Wire","x":125,"y":575,"x2":125,"y2":700,"height":0,"label":"nullnull","flags":0,"uid":"55499c20-8ce6-4af4-9112-5fe10f1a35dd","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":101,\"width\":16,\"x\":117,\"y\":475},\"second\":{\"height\":4,\"width\":4,\"x\":123,\"y\":698}}","segmentPoints":"[]"},{"name":"Wire","x":75,"y":350,"x2":125,"y2":350,"height":0,"label":"nullnull","flags":0,"uid":"3cd71f1d-f157-4236-b132-ebb9286ad2a3","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":32,\"width\":101,\"x\":-25,\"y\":334},\"second\":{\"height\":4,\"width\":4,\"x\":123,\"y\":348}}","segmentPoints":"[]"},{"name":"Wire","x":75,"y":700,"x2":125,"y2":700,"height":0,"label":"nullnull","flags":0,"uid":"0cc732b6-881e-4fc8-bc30-5fd138046268","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":24,\"width\":101,\"x\":-25,\"y\":688},\"second\":{\"height\":4,\"width\":4,\"x\":123,\"y\":698}}","segmentPoints":"[]"},{"name":"Wire","x":-25,"y":350,"x2":-25,"y2":700,"height":0,"label":"nullnull","flags":0,"uid":"bdb96223-d1d8-4e1c-9a92-7ba8f5bb2e86","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":32,\"width\":101,\"x\":-25,\"y\":334},\"second\":{\"height\":24,\"width\":101,\"x\":-25,\"y\":688}}","segmentPoints":"[{\"x\":-69,\"y\":350},{\"x\":-69,\"y\":700}]"},{"name":"Text","x":-143,"y":285,"x2":428,"y2":286,"height":0,"label":"nullnull","flags":0,"uid":"35247836-c54b-472a-bb8b-c43715481ce1","scoped_values":[],"text":"Turn the switch on and then off to start the oscillation","fontsize":25},{"name":"Post","x":125,"y":350,"x2":125,"y2":350,"height":0,"label":"nullnull","flags":0,"uid":"cbadb417-ee7a-477e-bcea-212a1daeacb9","scoped_values":[]},{"name":"Resistor","x":-25,"y":350,"x2":75,"y2":350,"height":0,"label":"R3","flags":0,"uid":"5843a8d2-0320-4135-84ba-248f12e4c86b","scoped_values":[],"resistance":10},{"name":"Inductor","x":125,"y":475,"x2":125,"y2":575,"height":0,"label":"L1","flags":0,"uid":"2c50be6b-b815-4d43-9da4-52f065017a91","scoped_values":[0],"inductance":1},{"name":"Capacitor","x":-25,"y":700,"x2":75,"y2":700,"height":0,"label":"C2","flags":0,"uid":"f207bfdf-826d-447d-855c-632a71a31cba","scoped_values":[],"capacitance":1.4999999999999999E-5,"initialVoltage":0.001},{"name":"Resistor","x":225,"y":700,"x2":325,"y2":700,"height":0,"label":"R3","flags":0,"uid":"70cb204c-c81c-41a6-9789-b73631571bda","scoped_values":[],"resistance":100},{"name":"Post","x":125,"y":700,"x2":125,"y2":700,"height":0,"label":"nullnull","flags":0,"uid":"52b8baa3-a57f-4b98-971c-8d38e3fe3614","scoped_values":[]},{"name":"Wire","x":125,"y":700,"x2":225,"y2":700,"height":0,"label":"nullnull","flags":0,"uid":"3e8877aa-2f7b-4299-96d1-a3ef5f4fd353","scoped_values":[],"blockingBBoxes":"{\"first\":{\"height\":4,\"width\":4,\"x\":123,\"y\":698},\"second\":{\"height\":32,\"width\":101,\"x\":225,\"y\":684}}","segmentPoints":"[]"},{"name":"DCVoltage","x":325,"y":600,"x2":325,"y2":500,"height":0,"label":"V3","flags":0,"uid":"6e327913-2148-4d7a-be1c-42feae575a67","scoped_values":[],"waveform":0,"frequency":40,"maxVoltage":5,"bias":0,"phaseShift":0,"dutyCycle":0.5},{"name":"Switch","x":150,"y":350,"x2":250,"y2":350,"height":0,"label":"SWT2","flags":0,"uid":"3c776e45-b471-4b59-9546-516b1998d731","scoped_values":[],"position":1,"momentary":false}],"label_map":"{}"}`

damiandaminczyk, 2022-11-21:
I can read also all types of comonents like output below, those strings should be hidden:
`return (b)(str.equals("GroundElm") ? new h(paramInt1, paramInt2) : (str.equals("ResistorElm") ? new f71(paramInt1, paramInt2) : (str.equals("CapacitorElm") ? new a(paramInt1, paramInt2) : (str.equals("PolarCapacitorElm") ? new rx0(paramInt1, paramInt2) : (str.equals("InductorElm") ? new j(paramInt1, paramInt2) : ((str.equals("DCVoltageElm") || str.equals("VoltageElm")) ? new hm(paramInt1, paramInt2) : (str.equals("ACVoltageElm") ? new d(paramInt1, paramInt2) : (str.equals("CurrentElm") ? new e(paramInt1, paramInt2) : (str.equals("NoiseSourceElm") ? new bs0(paramInt1, paramInt2) : (str.equals("PulseSourceElm") ? new u01(paramInt1, paramInt2) : (str.equals("SawtoothSourceElm") ? new ub1(paramInt1, paramInt2) : (str.equals("SquareSourceElm") ? new yi1(paramInt1, paramInt2) : (str.equals("TriangleSourceElm") ? new ct1(paramInt1, paramInt2) : (str.equals("SwitchElm") ? new nm1(paramInt1, paramInt2, false) : (str.equals("DiodeElm") ? new mq(paramInt1, paramInt2) : (str.equals("PostElm") ? new fy0(paramInt1, paramInt2) : (str.equals("LEDElm") ? new qg0(paramInt1, paramInt2) : (str.equals("NTransistorElm") ? new fq0(paramInt1, paramInt2) : (str.equals("PTransistorElm") ? new wv0(paramInt1, paramInt2) : (str.equals("PotElm") ? new gy0(paramInt1, paramInt2) : (str.equals("AmmeterElm") ? new n2(paramInt1, paramInt2) : (str.equals("VoltmeterElm") ? new u02(paramInt1, paramInt2) : (str.equals("PushSwitchElm") ? new y01(paramInt1, paramInt2) : (str.equals("SwitchSPDTElm") ? new om1(paramInt1, paramInt2) : (str.equals("TransformerElm") ? new rr1(paramInt1, paramInt2) : (str.equals("ACRailElm") ? new c(paramInt1, paramInt2) : (str.equals("RailElm") ? new m(paramInt1, paramInt2) : (str.equals("TextElm") ? new ro1(paramInt1, paramInt2) : (str.equals("ZenerDiodeElm") ? new u22(paramInt1, paramInt2) : (str.equals("FuseElm") ? new s40(paramInt1, paramInt2) : (str.equals("OpampElm") ? new lv0(paramInt1, paramInt2) : (str.equals("LogicGenElm") ? new ak0(paramInt1, paramInt2) : (str.equals("AndGateElm") ? new r2(paramInt1, paramInt2) : (str.equals("OrGateElm") ? new tv0(paramInt1, paramInt2) : (str.equals("XorGateElm") ? new p22(paramInt1, paramInt2) : (str.equals("XnorGateElm") ? new o22(paramInt1, paramInt2) : (str.equals("LogicOutElm") ? new bk0(paramInt1, paramInt2) : (str.equals("NandGateElm") ? new lq0(paramInt1, paramInt2) : (str.equals("NorGateElm") ? new gs0(paramInt1, paramInt2) : (str.equals("NotGateElm") ? new is0(paramInt1, paramInt2) : (str.equals("RelayElm") ? new b61(paramInt1, paramInt2) : (str.equals("BulbElm") ? new ta(paramInt1, paramInt2) : (str.equals("IC555Elm") ? new r90(paramInt1, paramInt2) : (str.equals("MosfetNElm") ? new hp0(paramInt1, paramInt2) : (str.equals("MosfetPElm") ? new ip0(paramInt1, paramInt2) : (str.equals("SparkGapElm") ? new ki1(paramInt1, paramInt2) : (str.equals("DCMotorElm") ? new gm(paramInt1, paramInt2) : (str.equals("TunnelDiodeElm") ? new ft1(paramInt1, paramInt2) : (str.equals("ADCElm") ? new e(paramInt1, paramInt2) : (str.equals("DFlipFlopElm") ? new im(paramInt1, paramInt2) : (str.equals("SchmittElm") ? new mc1(paramInt1, paramInt2, false) : (str.equals("InvertedSchmittElm") ? new kd0(paramInt1, paramInt2) : (str.equals("AMSourceElm") ? new f(paramInt1, paramInt2) : (str.equals("FMSourceElm") ? new bx(paramInt1, paramInt2) : (str.equals("LEDMatrixElm") ? new rg0(paramInt1, paramInt2) : (str.equals("SchottkyDiodeElm") ? new nc1(paramInt1, paramInt2) : (str.equals("AntennaElm") ? new k3(paramInt1, paramInt2) : (str.equals("TFlipFlopElm") ? new ym1(paramInt1, paramInt2) : (str.equals("SevenSegmentElm") ? new oe1(paramInt1, paramInt2) : (str.equals("SevenSegmentDecoderElm") ? new ne1(paramInt1, paramInt2) : (str.equals("BuzzerElm") ? new ya(paramInt1, paramInt2) : (str.equals("ProbeElm") ? new bz0(paramInt1, paramInt2) : (str.equals("WobbulatorElm") ? new v12(paramInt1, paramInt2) : (str.equals("OhmMeterElm") ? new zt0(paramInt1, paramInt2) : (str.equals("TransformerTappedElm") ? new sr1(paramInt1, paramInt2) : (str.equals("ThyristorElm") ? new cq1(paramInt1, paramInt2) : (str.equals("MultiplexerElm") ? new up0(paramInt1, paramInt2) : (str.equals("DeMultiplexerElm") ? new fn(paramInt1, paramInt2) : (str.equals("AdderElm") ? new r1(paramInt1, paramInt2) : (str.equals("JKFlipFlopElm") ? new ud0(paramInt1, paramInt2) : (str.equals("SequenceGeneratorElm") ? new ed1(paramInt1, paramInt2) : (str.equals("DiacElm") ? new up(paramInt1, paramInt2) : (str.equals("ClockElm") ? new if(paramInt1, paramInt2) : (str.equals("LDRElm") ? new pg0(paramInt1, paramInt2) : (str.equals("CounterElm") ? new fl(paramInt1, paramInt2) : (str.equals("SpeakerElm") ? new oi1(paramInt1, paramInt2) : null))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))));`

damiandaminczyk, 2022-11-21:
Please address those security issues in next version.

voltsimapp, 2022-11-22:
Regarding the unobfuscated code. we have reduced the obfuscation level in this update, which might have caused this. We will increase the obfuscation in the next update.

Regarding the data in the database embedded in the app. We do not consider this a security risk. We have an option of exporting a project as JSON in the timeline. Which would let the user see all this info anyway.

voltsimapp, 2022-12-10:
We have increased the obfuscation level back up in 0.2.0

closed 2022-12-10 by voltsimapp