VoltSim/VoltSim #23: Import netlist function

mh0520, 2022-11-23:
Are there plans to add the ability to import netlist or schematic files from other editors, like kicad?

voltsimapp, 2022-11-24:
@mh0520 Importing schematic from other editors is not in the pipeline yet. 
We would be focusing on improving the simulation speed and adding more components before moving on to more integrations.

gsingh93, 2023-06-26:
What about exporting a netlist so I can take a design from the app and open it in something like KiCad, so I can add footprints and layout a PCB?

voltsimapp, 2023-08-18:
This can be done, as internally the component parameters are similar. Currently, the app saves the circuits in json.  This is in the pipeline but not a high priority at the moment.

voltsimapp, 2023-12-21:
Closing this as this is not a priority. Will reopen when we add this to our timeline again.

closed 2023-12-21 by voltsimapp