VoltSim/VoltSim #26: Bugs when flipping IC components

jacius, 2022-11-30:
Using VoltSim 0.2.0 on Android 11, there are bugs which affect horizontal and vertical flipping with all integrated circuit components. At least the following components are affected:

* Led Matrix
* 555
* DAC
* ADC
* D FlipFlop
* T FlipFlop
* JK FlipFlop
* BCD to 7Seg
* 7 segment
* Mux
* DeMux
* Adder
* Seq Gen
* Counter

There are two bugs:

1. The components revert to their original orientation after saving and reloading the project. HFlip and VFlip are both reverted. This affects the behavior as well as the appearance of the component.
2. The VFlip button is enabled (not dimmed) but has no effect on ICs which do not have terminals on the top or bottom. This affects all of the above except Led Matrix, 555, Mux, and DeMux. The VFlip button should be disabled if it will have no effect.

Also, I am not sure if this is a bug or intended behavior, but HFlip only affects terminals on the left/right sides, and VFlip only affects terminals on the top/bottom sides. In other words, HFlip does not reverse the order of terminals on the top/bottom sides of an IC, and VFlip does not reverse the order of terminals on the left/right sides of an IC. This is closely related to bug 2 above.

voltsimapp, 2022-12-21:
We are working on a more exhaustive IC layout editing feature that can be reused for creating custom ICs too. We are planning to have it implemented for next release. So it should fix all these issues.

voltsimapp, 2023-08-31:
We have reworked the flipping for all components. We have disabled IC flipping for now until the IC editing is releaesd.

voltsimapp, 2023-12-21:
All the flipping issues are resolved. The IC flipping will be enabled with the custom IC editor implemented tracked in https://github.com/VoltSim/VoltSim/issues/21

closed 2023-12-21 by voltsimapp