VoltSim/VoltSim #27: inverted mosfet symbol

Awen2345, 2022-12-02:
[image]
The one on the far left is an n channel mosfet but the symbol is like a p mosfet And secondly To the right of the p mosfet the symbol is like the n mosfet So I can't tell the difference between n mos and pmos

Awen2345, 2022-12-02:
[image]
This should be the correct symbol

voltsimapp, 2022-12-03:
You can flip it to the correct position by the VFlip button in the bottom menu. 
Will update the default orientation in the next update.

voltsimapp, 2023-02-19:
Fixed in 0.2.1, please reopen if the issue still exists.

closed 2023-02-19 by voltsimapp