VoltSim/VoltSim #29: Feature request: Element type on select

Hbaldwin, 2022-12-10:
It would be really nice for newbies like me if when i selected a component a label with the type of component was ma lde visible. Possibly right above the toolbox bar on the bottom.

voltsimapp, 2022-12-10:
Hi @Hbaldwin , Currently if you enable the multimeter it would show the name of the selected component along with some other info about the component. 

[image]

Was this what you were looking for? 
If so, maybe we can improve the UX so that this is more obvious. Enable this by default maybe?

Hbaldwin, 2022-12-10:
That is very much what i was looking for. It would be nice to be able to
just have the name. But this is great and very helpful. Feel free to close
this request. I tgink default enabled might be useful but by no means
neccesary. Thanks for your help, and work on such a great project!!!!

Hbaldwin, 2022-12-11:
The multimeter can accomplish this well, though it takes up a bit more space than neccesary for just the name. Closing for that reason.

closed 2022-12-11 by Hbaldwin