VoltSim/VoltSim #31: Export Speaker buffer as WAV file

jacius, 2022-12-16:
It would be really cool if it were possible to export the data from a Speaker's buffer as a WAV sound file. This would enable VoltSim to be used as a simple synthesizer to create ringtones, game sound effects, and more. There are other improvements that could be added later, such as higher maximum buffer size, configurable bitrate / sampling rate, or FLAC/MP3 export. But a basic WAV exporter would be a great first step, and very useful.

The workflow I am imagining is:

1. Run the simulation until the Speaker's buffer is full.
2. Open the sidebar and click on Export, which is currently used to export the circuit as an image.
3. The Export dialog would still have a button for exporting the image, but would also have a button for each Speaker in the circuit, with the Speaker's label. If a Speaker's buffer is not full, that Speaker's button could be grayed out and "(not full)" appended to the label to help users understand what they need to do.
4. Clicking the button for a Speaker would open the window to choose where to export the file, same as when exporting an image.

voltsimapp, 2022-12-19:
Thank you for the detailed suggestion. We have added this to our pipeline and it is planned for the next release. 
Currently, I have planned a slightly different flow. when a speaker is selected, we will show an export menu in the bottom menu bar. 

We have also considered other suggested features like "maximum buffer size", "configurable bitrate / sampling rate" as they are low-hanging fruits.

jacius, 2022-12-19:
Sounds great, thank you!