VoltSim/VoltSim #32: Simulator Si Laggy

Awen2345, 2022-12-20:
I wonder why the Simulator suddenly crashes when using 2 oscilloscope plots

voltsimapp, 2022-12-20:
That should not happen. Can you please share the circuit which is causing the crash?

Awen2345, 2022-12-20:
https://voltsim.page.link?link=https://www.voltsimulator.com?id=Iv6UcMsmF9rqTNnFA99p&apn=com.fallenbug.circuitsimulator

Awen2345, 2022-12-20:
https://user-images.githubusercontent.com/87870432/208772306-0b93acdf-985c-47ec-a6ca-38a96d18557b.mp4
for some reason the Simulator slowly crashes like the oscilloscope slows down that's the bug

voltsimapp, 2022-12-20:
Looks like a memory leak, thank you for reporting this.  I am unable to reproduce this on the test device. Can you please provide the following info?
App version:
Android version:
Device make and model:

Awen2345, 2022-12-21:
App Version : 0.2.0
Android Version : Android 9
Device Model : OPPO CPH2089

voltsimapp, 2025-05-04:
Should be solved in new version, please reopen if the issue still persists. 
Released in 0.3.7

closed 2025-05-04 by voltsimapp