VoltSim/VoltSim #33: Tema escuro

ProfetaDJ, 2023-01-08:
Ícones na barra inferior do projeto ficam ocultos quando ativado o tema escuro do aparelho. 
Sugestão: Alterar cores dos ícones e/ou criar temas específicos.

ProfetaDJ, 2023-01-09:
[image]
[image]
[image]

voltsimapp, 2023-01-09:
Thank you, we will fix this in the next release.
Obrigado, vamos corrigir isso no próximo lançamento.

nithinmurali, 2023-02-11:
@ProfetaDJ  Você pode fornecer algumas etapas para reproduzir esse problema? Não podemos reproduzir isso do nosso lado.

Can you please provide some steps to reproduce this issue? We are unable to reproduce this on our end.

voltsimapp, 2023-02-11:
@ProfetaDJ Você pode fornecer algumas etapas para reproduzir esse problema? Não podemos reproduzir isso do nosso lado.

Can you please provide some steps to reproduce this issue? We are unable to reproduce this on our end.

voltsimapp, 2023-05-17:
We are unable to reproduce this and since there is no response from the reporter, we can closing this issue. Please feel free to reopen if you face this issue again.

closed 2023-05-17 by voltsimapp