VoltSim/VoltSim #41: DC Current and Inductor only  in series not working properly.

MohammadAbdulGafoor, 2023-05-20:
When DC current source in series with Inductor , if I select Inductor for readings and when I select Current source , then Current flow stops.  When I select other element in circuit and then select Current source the current flow starts.  Graphing is also now working properly for this combination(DC current source in series with Inductor only).
Graph change when I select components in circuit one after other.

voltsimapp, 2023-05-20:
Thank you for reporting this bug. We are unable to reproduce this issue on our end. can you please share the circuit where we can reproduce this (use the share option on the projects screen)?  or maybe a screen recording?

MohammadAbdulGafoor, 2023-05-20:
https://voltsim.page.link?link=https://www.voltsimulator.com?id=hjVKqbkjC3ulMKW2tMx5&apn=com.fallenbug.circuitsimulator

voltsimapp, 2023-05-20:
Thank you! we can reproduce the issue and will be fixed in the next update. It's a minor fix.

MohammadAbdulGafoor, 2023-05-20:
Am I going to get any Reward such as free subscription to the app. Just
asking. I am a kid , I don't know anything much about this.

voltsimapp, 2023-05-20:
:smile:  Sure! please email voltsim.app@gmail.com mentioning this thread, we will have something for you :)

MohammadAbdulGafoor, 2023-05-20:
> 😄 Sure! please email ***@***.*** mentioning this thread, we

Thank you very much

closed 2023-05-21 by MohammadAbdulGafoor