VoltSim/VoltSim #48: Capicators bug

srantheman, 2023-11-21:
I cant plug capicators positive with positive, and negative with negative

srantheman, 2023-11-23:
Add power configure on zener diode

srantheman, 2023-11-23:
> Currently, all capacitors are ideal. So connecting them in parallel
> without any resistance could result in infinite oscillations. We currently
> throw an error when capacitors are connected in parallel to prevent this.
> To fix this please add a small resistor in series to the capacitors.
> Currently from the edit menu of the Zener diode, you can change the
> breaking voltage. Is this what you are looking for?

I cant get inpulse from capators

voltsimapp, 2023-11-23:
Can you please share a project where the circuit is not behaving correctly? (use the share button to create a link)

srantheman, 2023-11-23:
https://voltsim.page.link?link=https://www.voltsimulator.com?id=wtupZtlpEFd0Tc8QlumI&apn=com.fallenbug.circuitsimulator

srantheman, 2023-11-23:
Thyristor :
model - КУ202Н
repetitive peak off-state voltage - 400 V
on-state current - 10 A
gate trigger current - 100 mA
gate trigger voltage - 1.5 V
operating temperature - -60 C to +85 C
mounting - panel, M6.

srantheman, 2023-11-23:
More thrystor specification
https://asenergi.com/en/products/thyristors-t-stud/t122-20-4.html

voltsimapp, 2023-11-23:
We have a Thyristor component. But I am not sure if we expose enough parameters to simulate the linked model. We will look into this and get back.

srantheman, 2023-11-23:
https://lt.wikipedia.org/wiki/Stabilitronas

voltsimapp, 2023-12-21:
Hello, in the latest update 0.2.8 you can connect capacitors in parallel.