VoltSim/VoltSim #49: project storage

ct1igu, 2023-11-23:
Hi,

Firstly,  Voltsim is fantastic I use it alongside LTSpice & QSpice.

2 questions:

Is there a way of moving a project from one Android tablet to another? 
What is the location of the saved project files?

Thanks for any help.
Rod

voltsimapp, 2023-11-23:
Hi, 
You can use the share option to export a project from one device to another. The share option will provide a link and you can open the link in another device to import the project. 

The projects are not saved as files but are rather saved in the internal database. We save the projects in a custom JSON format. Is there some specific use case for accessing the saved files?

MightyShadow702, 2023-11-23:
Hey there, I had the same question, so I would answer in my use case. For me, this would be for backup purposes, so if I lost my phone or how OP sad, for send it to a other device nearby, which is not conection to the Internet or something like this, this would be a genuines feature, for send just a file, via Bluetooth for example.

So, we had just more than just one option to share our projects or store them.

I would guess there are many other examples, too.

Best regards

MightyShadow702, 2023-11-23:
> Hi,
> You can use the share option to export a project from one device to another. The share option will provide a link and you can open the link in another device to import the project.
>
> The [...]

Hey there, I had the same question, so I would answer in my use case. For me, this would be for backup purposes, so if I lost my phone or how OP sad, for send it to a other device nearby, which is not conection to the Internet or something like this, this would be a genuines feature, for send just a file, via Bluetooth for example.

So, we had just more than just one option to share our projects or store them.

I would guess there are many other examples, too.

Best regards

voltsimapp, 2023-11-23:
Hi, 

That makes sense, There is already an export/import feature that we use internally. But that's raw JSON and is not open to the public. 

Having control of saved files helps us migrate data easily if we update the saving format. Also giving editable access to the files could mean there can be hard-to-debug crashes if the user edits some values in the saved files. 

But if this is a useful feature, we can look into ways of file sharing without making it editable and add some versioning.

ct1igu, 2023-11-24:
Hi,

Thanks for your rapid reply.

Works fine,  I  open the url in a browser which then suggests using the VoltSim app to import/use the circuit simulation. However, if I take the receiving machine offline the import/install  does not work so ....

Using a URL is an interesting way of transferring files between local machines. It begs the question, do you store the circuits/simulations on your servers?
Most of my data is on NAS so it would be great to just copy some files.

I'm using VoltSim on a Samsung tablet (small screen) but also in an Android emulator on my PC (large screen)

I would be interested to know more about in the url transfer method.

Regards

Rod

voltsimapp, 2023-11-26:
Hi, All the circuit data is stored locally. But when you share a circuit, it is uploaded to our servers and a link is created to the data. The default retention policy is 90 days. So the link could stop working after 3 months. 

There is also plans on implementing the whole workspace export by URL, but is not a priority at the moment. 

Okay, I will add file export/import to the feature pipeline. It could also be useful for users with more privacy concerns.

ct1igu, 2023-11-26:
Hi,

Many thanks for the explanation. 

I agree there may be privacy and intellectual property issues. It's just educational/hobby in my case but I look forward to the file export/import feature.

Thanks,

Rod

voltsimapp, 2023-12-21:
Hello, we have released the project importing and exploring as files in version 0.2.8

ct1igu, 2023-12-22:
Hello again,

Many thanks for this change, it works very well and I also like the 'interact' feature.

Wishing you much success.

Rod

closed 2023-12-30 by voltsimapp