VoltSim/VoltSim #56: Variable Capacitor

TheKezStrel, 2024-01-29:
To add Variable Capacitor Component.

Interested in using these in radio circuits and seeing how the frequency changes.

voltsimapp, 2024-01-29:
Hello,  thank you for the feedback. Currently, you can choose any capacitor and change its capacitance using the Quick-edit wheel when in `QEdit` mode. 

Is there any scenario that this does not cover or is there any reason you believe this won't work in your use-case?

TheKezStrel, 2024-01-29:
Understandable you'd not want to change what works. Just thinking of
newbies and that I couldn't find the variable capacitor symbol as I was
trying to build a radio tuning circuit.

TheKezStrel, 2024-01-29:
That was what I was thinking.

Anyting for simplicity.

I'll open another one for another idea of components that would be good to be added.