VoltSim/VoltSim #57: Change to transformers?

TheKezStrel, 2024-01-29:
I wonder if it would be possible to make a change to Transformers. They're currently set to use a ratio between coil N1 and N2. Most self-build hobbyists like myself recognize coils in terms of turns. Perhaps a change to the dialogue allowing the user to input numbers of turns to the coil in the transformer would be beneficial. If the ratio between the coils is necessary for the calculations in the back end, then on the close of the dialogue the ratio should be calculated to be used in the simulation.

Another related component addition would be similar to the centre tap transformer, but having dual outputs with differing turns. I have seen these in circuit diagrams and multi-voltage systems. (n1 primary, N2 and N3 secondaries that have differing turns for different voltage outputs off the same primary).

voltsimapp, 2024-01-30:
Sure, We can add a second input field for turns in N1 and default it to 1, which will take the ratio of both inputs in the backend. So user should be able to input ratio or turns as desired. This is a small change and you can expect it in the next update.

Making the tapped transformer more customizable is a planned change, and is in our pipeline.

voltsimapp, 2024-05-02:
Fixed in 0.2.13; Please reopen if the issue is still not resolved.

closed 2024-05-02 by voltsimapp