VoltSim/VoltSim #60: Show name of component somewhere after placing

marnovdm, 2024-02-21:
Sometimes in example projects I don't recognize an icon and the only way to resolve that to a name (as far as I know) is by browsing the menu you get when you add a new item. Instead of 'Edit Element' a title like 'Edit LED' or something would be a small but imho very useful improvement.

marnovdm, 2024-02-21:
Thanks that's awesome!

1. I had no issues locating the multimeter, I just hadn't noticed the component name there.
2. Assuming you mean top right buttons here for multimeter? Otherwise I have clearly had issues locating them as I don't see any toggle buttons top left :-)

I did have some issues connecting components, it doesn't always seem to connect when just placing 2 things next to eachother, and sometimes it is impossible to connect them with a wire if they are only separated by one 'block'. When you move it to at least 2 blocks between them it works as expected so not a big deal.

voltsimapp, 2024-02-21:
> it doesn't always seem to connect when just placing 2 things next to each other,

Do you mean even when nodes of two components are overlapping it's not getting connected? Can you please provide an example so that we can reproduce this?  you can create a project with this bug and share the link here.

> sometimes it is impossible to connect them with a wire if they are only separated by one 'block'.

Thanks for reporting this! will fix this in the next update.

marnovdm, 2024-02-21:
Obviously when I try to reproduce it now I can't :-) I'll try some more and report back here when I figure out steps to reproduce. It could also have been a case of me trying to connect a component to the middle of a wire which looks like a connection but does not work. It might be nice if that was possible.

voltsimapp, 2024-02-21:
For connecting to the middle of a wire, you can select the wire and split it (bottom menu). this would add a node (join) to the wire. You can then move the node to where you want to connect more components to.

voltsimapp, 2024-05-02:
Fixed in 0.2.13; Please reopen if the issue is still not resolved.

closed 2024-05-02 by voltsimapp