VoltSim/VoltSim #61: Please add DPDT switch.

derei, 2024-02-25:
I had to make a circuit with a DPDT switch, and the only option I had, was to customize a relay, as the only solution.

voltsimapp, 2024-02-28:
Hi, this feature is already developed. You can expect this in the next release.

derei, 2024-04-04:
Quick follow-up on this request: I understand major updates take time, but incorporating essential components like the DPDT switch could significantly enhance usability, especially for educational purposes.
Recently, I've been teaching students about DC motor polarity reversing. Simulating a DPDT switch using two SPDT switches has been somewhat confusing for them. Adding a few basic components, such as the DPDT switch, in a smaller patch release would be incredibly helpful.

Thanks for considering this, looking forward to any potential updates!

voltsimapp, 2024-04-05:
Hello @derei , 
Thank you for bringing this to our attention. We understand the importance of incorporating features like the DPDT switch, especially for educational purposes, and we appreciate your patience as we were working on a major update.

As requested, we will have a small release in two weeks with the DPDT switch and a few other minor updates.

voltsimapp, 2024-05-02:
In this release, we have added the DPDT switch as a new component and have also added grouping for controlling multiple SPST switches together.

closed 2024-05-02 by voltsimapp