VoltSim/VoltSim #67: VCVS returns to default value

Geitinisveitinni, 2024-04-07:
VCVS returns to default value if you close and reopen a project or use the undo button

voltsimapp, 2024-04-08:
Hello, can you please provide the steps to reproduce this issue? Some screenshots or screen recording would be helpful as well.

voltsimapp, 2024-04-08:
Thank you for the video and for sharing the project. Unfortunately it's still not clear to me what is not getting saved. The equation before and after saving is the same "5*(a-b)"  ?

Geitinisveitinni, 2024-04-08:
Yes but the voltage and current values everywhere in the circuit are wrong until I press edit and save

voltsimapp, 2024-04-10:
Thank you for reporting this bug! We will fix this in the next release.

voltsimapp, 2024-05-02:
Fixed in 0.2.13 ; Please reopen if issue still persists

closed 2024-05-02 by voltsimapp