VoltSim/VoltSim #76: Add Comparator, LM13700, and VC Analog Switches

cjmventer, 2024-07-12:
I work in synthesizer and musical instrument design and love the app for sketching out ideas for modules, pedals, and rack units. However, there are a few key missing parts that we use very often in this field that would really help. Falstad has all of these elements, for example:

OTAs (such as LM13700) are used all the time  in synthesizers and other audio circuits for creating VCAs and filters among other things. This would be very good to have.

A basic open collector comparator should be added (such as LM311 or LM319).

Voltage/logic controlled analog switches would also be very useful to have. This doesn't necessarily need to be based on a particular chip, but even just a make or break bi-directional switch model with an input pin to provide hi or low voltage inputs and equivalent on or off resistance values to edit would suffice for a lot of applications.

Thanks for your time!

voltsimapp, 2024-07-15:
Thank you for the suggestions. OTA and analog switches will be added in the next update.

voltsimapp, 2024-08-17:
In the latest update (0.2.14) we have added analog switches. OTA is planned for the next release.

cjmventer, 2024-08-17:
Awesome! I see there is a comparator also, but I seem to be having trouble getting it to work. Leaving the output unconnected or adding a voltage source with a limiting resistor to the output (for hi-z/gnd output) did not seem to result in changing voltage at the output in response to the + and - input voltages.

voltsimapp, 2024-08-18:
Hello @cjmventer , Thank you for notifying this. Looks like that bug slipped past the tests.  Will send a update with the fix soon.

voltsimapp, 2024-08-18:
Fixed in patch release 0.2.15. also added OTA

voltsimapp, 2024-08-20:
closing this issue as all components are implemented. Please re-open if you find any more issues.

closed 2024-08-20 by voltsimapp