VoltSim/VoltSim #79: Reseting relay properties

SuperRelay, 2024-08-17:
Hello, great app you have made
I see tons of potencial in it, it looks great in mobile!
But i see a very big problem
Some properties when you edit a relay are just not good, i can change them, but even if i save (tried about 30 times) they change back to the defaults.
Resistance OFF
Operating current
Release current
The other properties look fine, please take a look at that, if you fix it them i am sure i will buy full version from you, you have done a great work, it looks easy to use, you just need to know how to make an electrical drawing to understand this.
But of course if you need to change properties every time you go back to a project them it is unusable.
Of course it will also need tp have relays of at least 2 contacts
Thank you very much

voltsimapp, 2024-08-18:
Hello @SuperRelay , This a known bug and we have fixed this in the latest version (0.2.14). Its currently in open testing and will be released soon. In the relay you can set the number of contacts but its a premium feature [any extension] so it's not visible by default.

voltsimapp, 2024-08-20:
Fixed and released in 0.2.15

closed 2024-08-21 by voltsimapp