VoltSim/VoltSim #80: Not exporting to .pdf

SuperRelay, 2024-08-28:
I am trying to export to .pdf a project, but it seems its not possible
Other projects look fine

voltsimapp, 2024-08-28:
Are you not able to export any projects to just one project? What is happening when you try to export to pdf?

SuperRelay, 2024-08-28:
It is just one project, it quits the project that is open, and it creates only one .pdf with 97 bytes

voltsimapp, 2024-08-28:
Please share link to that project [use share button in projects screen].

SuperRelay, 2024-08-28:
Of course, here it is
https://www.voltsimulator.com/dl?id=QTSKdEZtNfH62p66mEEf

voltsimapp, 2024-08-28:
Hmm, looks like the image is too big and is taking some time to render (probably high memory). when tested locally, it creates a pdf of around 5MB. probably the limitation could be your divide memory.

Do you see an "image exported " message in the app, or does it quite unexpectedly? 
Can you try exporting to an Image? 
You can also try moving around the components so that total file size is reduced.

[Pro AD 0.0.2-light.pdf](https://github.com/user-attachments/files/16789038/Pro.AD.0.0.2-light.pdf)
[Pro AD 0.0.2 (2).pdf](https://github.com/user-attachments/files/16789040/Pro.AD.0.0.2.2.pdf)

SuperRelay, 2024-08-28:
Ok, i do not see the usual "image exported" message, and it just quits to the selection of projects screen.
And i can export to an Image with no problem.
If i reduce the size like you suggested then i can export it to .pdf, but it does not look like memory issue, at least on the device side, i tried to increase the ram memory (extended memory of course), and also just tested cleaning some files and getting more 20gb free of storage, and the results were the same, if i increase (space out) the size by just a few lines, it did the same problems anyways

voltsimapp, 2024-12-05:
@Awdyorebel  are you facing the issue with all the circuits? what about a circuit with just 2 or 3 components?