VoltSim/VoltSim #82: Multi select isdue

Biggidy, 2024-09-04:
When I multi select components it doesnt drag the lines/cables - they stay still despite being highlighted.

voltsimapp, 2024-09-07:
Hi, curently if you move components all the wires between them will move automatically. Are you trying to move just the wires? why not just connect them after moving? Can you show a screen recording of what you are trying to achieve?

Biggidy, 2024-09-07:
Hi,

Thanks for your response.

It is when I duplicate sections of a circuit or highlight large parts of a
circuit to move parts.

Please see video attached.

Kind regards

Damian

voltsimapp, 2024-09-07:
Hello,

It looks like you missed attaching video? If you have trouble attaching
video in github please send it to ***@***.***

Biggidy, 2024-09-07:
Please find attachment

https://github.com/user-attachments/assets/05154659-b167-458f-b3c2-765484103365

voltsimapp, 2024-09-08:
Thank you, that definitely looks like a bug and we can reproduce it. We will be fixing it in the next update.

It looks like the issue only happens when the simulator is stopped. If the simulator is running and you duplicate and move the components the wires also move with them. Can you confirm?

Biggidy, 2024-09-08:
Yes, that is correct, only when the simulation is stopped. 

Thanks for pointing it out, now I can work around the issue until the update 👍🏻

voltsimapp, 2024-09-30:
This bug is fixed in the latest update 0.2.16

closed 2024-09-30 by voltsimapp