VoltSim/VoltSim #83: Regarding Extension

chmosiii, 2024-09-09:
New User : I like the app, however something doesn't seem to work ..

I paid on Sept/04/24 for the scope extension. The scope got updated as intended and parts in the parts library that were blocked got enabled too which I didn't expect to happen. However many parts are only symbols and cannot be simulated yet at the moment.

So I clicked on the EXTENSION symbol on the dash board to see if there were extra payments to make to get the parts that cannot be simulated yet and I clicked it thinking the parts would now run, but nothing happened except I believe I made a second payment of $12.64 that passed in my bank account waiting to be charged for nothing.

Some clarifications are needed ..

Regard !!!

voltsimapp, 2024-09-09:
Hi, all the components are simulated. If you have purchased the "Pack extension" [the $12.64 one]. All the features should be unblocked and all components should work.
Can you please share a screen recording of the components/parts not working?
Can you also please share your purchase id from google play? we will check for any permission issues there.

chmosiii, 2024-09-10:
Hello,

I guess I'm in trouble. When I clicked the payment links, my PayPal link opened and I accepted to make the two payments (The Instrument Extension & The Extension Pack). However the payment was made outside Google Play Store, therefore VoltSim doesn't appear nowhere on my Play Store purchases area. I then checked the email I registered on your site, no bill there either. I then checked on my PayPal account, again here nothing shows I made a purchase on Voltsim althoughI paid through paypal. The scope was updated as I mentioned previously and the grayed parts as well, but dome are not simulable.

Speaking of the parts I can't simulate, one of them is the CRYSTAL. When I click to edit the frequency I want it to oscillate, all I see is the label and nothing else. I can't tell it to the frequency I want it to oscillate through the EDIT area. I drew a simple crystal oscillator, it does oscillate, but very erratically and nowhere I can't Edit the frequency of oscillation. As I said, when I click the Edit button, all I see is the crystal label and nothing else. There are other parts that aren't simulable, such as  the OpampReal I only see the label and cannot edit amps parameter in it.

As I mentioned above, I purchased the Instrument Extension that worked fine and that enabled all the grayed parts, and today the Extension Pack ($10.99).

I dont mind if I lost the $10.99 Extension Pack payment providing it passed, I just want to be able to simulate the parts.

Best regards ..

voltsimapp, 2024-09-10:
Hi, Looks like the extension pack is active. 

Unfortunately, you cannot edit the frequency of the crystal oscillator. We will add a feature request for this and will fix this in the next update.

The Opamp-real is internally IC741 so its parameters cannot be changed as it matches that of 741. you will be able to change this model in future releases.
If you want to use an Opamp whose parameters can be changed, please use Opamp (Ideal) [The Opamp without rails]. There you should be able to update gain and rail voltages.

Let us know if you are missing any other parameters for simulation.

chmosiii, 2024-09-10:
Ok great !! 

Now I understand ..

Regard ..

voltsimapp, 2024-09-30:
Ability to configure crystal frequency added in 0.2.16

closed 2024-09-30 by voltsimapp