VoltSim/VoltSim #86: Minor niceties request

Explrr, 2024-10-26:
Using version: 0.2.16
I would like to suggest a few niceties. 

1) When splitting a wire, it would be nice if the new wire had the same color as the original wire before it was split. 

2) When attaching new wires to other wires, or to components with wires attached, it would be nice if I had to option to have the new wire color change to the same color as the wire I'm attaching to. 

3) There are default colors that cannot be changed. Such as the voltage color for positive is GREEN and negative is RED. (In my projects, the solderless PCB uses RED for positive and BLUE for negative) When I change the wire colors to reflect my projects, the voltage colors stay the same. It would be nice if I had the option to change the default colors of the voltage. 

4) When a component isn't grounded properly, it displays a barely noticeable color difference. It would be nice if I could change the color, hue, and/or intensity to make it more visible. 

5) When I make a change to a schematic then decide to go to the main menu, I am asked if I want to save my changes. Regardless if I select Yes or No, the app saves my changes. It would be nice if I choose No and my changes weren't saved. 

6) It would be nice if I could order my list of projects based on date or name. Ascending or descending.

voltsimapp, 2024-10-27:
Thank you for your feedback.

1, 3 sounds good, will add to our feature pipeline.

2: If you can choose multiple wires and change all of their colors at once. Will that solve this issue?

4: can you please share a screenshot showcasing this?

5: Do you have autosave enabled in the app settings? Can you verify if the project is getting saved even after autosave id disabled?

6. This feature is in the works, and you can expect it very soon.

Explrr, 2024-10-27:
2: That would be a nice option too.

4: This appears only when the option Show Voltage is On.
Improper Ground.png, Proper Ground.png uploaded.

[image]
[image]

5a: Ah, I forgot about this option. Now that makes sense. It appears to work with it off.

5b: In that case, with Autosave On, why does it ask if I want to save changes?

voltsimapp, 2024-10-29:
4. I am still a bit confused on what the bug is. Current behaviour is that any unconnected component or grounded component has "normal" component color which is gray. If the component is between -5V and 0V it would be a gradient of red. between 0 and 5V it would be a gradient of green. 

5b. Good point. The only scenario where this could happen is when the autosave is triggered when the "save confirm" dialog is being shown. I will check if this is indeed another bug.

Explrr, 2024-10-29:
4. I am not saying this is a bug. I am saying it is barely noticeable. At least on my tablet. I like the gradience, I just thought it would be nice if it was more visible.