VoltSim/VoltSim #90: OTA Voltage Rails and Iabc Voltage Bug

cjmventer, 2024-11-25:
Hi! I am very happy that OTAs have been added as they are very important to many types of circuits. However, I noticed a couple of issues with them.

1. When you change your voltage rails to something other than the default +/-9V, (for example +/-12V) the voltage at the Iabc pin stays around -8V instead of changing in relation to the new negative rail value. This seems to happen both  with a lower or higher negative rail setting.

2. For some reason, you also can't set the negative rail to 0V.  This should be allowed as plenty of OTA circuits run on a single positive supply voltage and use a bias Voltage at the input.

cjmventer, 2024-11-25:
[Screenshot_20241125_143029_VoltSim](https://github.com/user-attachments/assets/7d8f270d-d24c-490f-871b-52104074c225)

cjmventer, 2024-11-25:
You also get identical output results regardless of voltage rails, so I would assume the rails inputs are simply not working and always stay at +/-9V
[image]

voltsimapp, 2024-11-26:
Thank you for the detailed bug report. There indeed was a bug that was preventing values from getting updated internally.  Will release the fix in the next update.

voltsimapp, 2025-01-09:
fixed in 0.2.17

cjmventer, 2025-01-09:
Thanks so much!

closed 2025-01-27 by voltsimapp