VoltSim/VoltSim #91: Add arcs to overlapping wires not connected

rnbpsp, 2024-12-05:
Please add arcs to overlapping wires not connected

voltsimapp, 2024-12-05:
Is this regarding the wire crossing? Currently, the wires that are connected have a circular to denote that the joint is connected. Isnt that sufficient to differentiate between connected and unconnected crossing?

cjmventer, 2024-12-13:
I agree that having an arc where unconnected wires cross would make things clearer to read. Especially if sharing a screenshot of a schematic with someone, it can get a bit unclear.

DadoTC, 2025-02-18:
I don't agree and would prefer it to stay as it is now

nocandopdx, 2025-03-09:
> I don't agree and would prefer it to stay as it is now

I agree with this. If you were to add arcs, you would go from all connections being circles, to having circles where wires split, circles on components, but arcs where wires overlap. That is a step back when it comes to consistency and clarity. Arcs make sense for hand drawn circuits because they require the least amount of effort to draw. We're here to get away from drawing : )