VoltSim/VoltSim #92: Button bugs

nocandopdx, 2024-12-18:
I keep getting bugs with my buttons where all of s sudden they stop functioning or latch to an opposite state of what they're set to. Like a broken button in real life. Sometimes clicking edit on the button and just pressing okay restores functionality but it's a hit and miss. It will either work right away on the first try, but if it doesn't, repeating the edit hasn't worked. Restarting the app does not help. Stopping and resetting the simulation also does not help. This has happened on multiple projects, on two Android devices. I've tried uninstalling and reinstalling but the bug randomly comes back. Sometimes changing the simulation settings also gets it to start working, but the bug itself is not tied to a certain setting. So far I've ran into the bug on circuits with relays and noticed that selecting a relay coil while this is happening "refreshes" the glitched buttons, but the toggle remains nonfunctional. Almost like clicking the relay coil turns into a signal that activates the button state when they stop working. I have experienced the glitch at a half dozen different speed/time step/current speed settings. With regular switches and push buttons. Intermittent glitches are the absolute worst to troubleshoot but I really hope someone decides to look into this, I hope i provided enough information to figure out why this might be happening.

voltsimapp, 2024-12-18:
Thank you for reporting this issue. Can you share a video recording of the bug to better understand the issue? Sharing the project link would also be helpful (use the share button in the projects screen).

voltsimapp, 2024-12-20:
Thank you for sharing the video and examples. In the video, it looks like the button is functioning as expected? The push switch will bounce back to the off state when the button is released. And if not a push switch it will stay at that state.

nocandopdx, 2024-12-20:
This was not about the animation, sorry I guess that was not obvious. Nothing happens when it is pushed. it is suppose to send the current through the wire. When I jump the button terminals, with the wire, the current flows. When I push the button, it doesn't. Also when I switch to the latch button, same thing can be observed. When the relay is clicked on a different circuit, the current starts flowing. Let me know if you want me to take another video comparing the bugged and non bugged behavior.

voltsimapp, 2024-12-21:
Thank you for the explanation. That was indeed unexpected behavior. Looking at the circuit, when the switch is turned ON, the capacitor's reverse voltage increases significantly and occasionally triggers the error: "Max reverse voltage reached." It seems the capacitor might be getting shorted through the diode.

Since the simulation uses an ideal capacitor by default, shorting the capacitor (or any voltage source) could cause the simulation to either fail to converge or take much longer to do so. This, in turn, leads to the simulation lagging, which is likely what’s happening here.

**Solution:** Add a small internal resistance to the capacitor [screenshots attached] so that the cap is not ideal anymore. 

[image]
[image]

voltsimapp, 2024-12-21:
Let us know if this resolves the issue. We’ll work on providing better warnings to help catch cases like these in the future.

nocandopdx, 2024-12-21:
Thank you for looking into this and finding out what's going on. What you are saying is consistent with my experience, the issue came up in circuits that discharged capacitors via various methods and definitely lag that i thought had something to do with simulation speed settings. I also have encountered the error message you mentioned, but instead of adding internal resistance i increased reverse voltage. I had no idea this was what caused the lag 🫠

voltsimapp, 2025-05-04:
Released in 0.3.7, improved notifications whern simulation is lagging

closed 2025-05-04 by voltsimapp