VoltSim/VoltSim #93: What's going on with this relay

nocandopdx, 2024-12-26:
I think i might have come across another bug or maybe just something I don't understand. Why is the relay coil not responding to the pulses? The two NPN transistors with collectors in series with the relay coil are behaving normally, so why is the relay coil itself not doing anything when the transistors are switched on and off? Relay settings are all default. Ty

https://youtube.com/shorts/YSfc966fEHk?si=RsUGXZqDWS85sJ-t

https://www.voltsimulator.com/dl?id=cQV2dHDeM2xcUsETqXyn

cjmventer, 2024-12-27:
It seems like the LED resistor is too high for the transistors to pull down the coil to a low enough voltage to deactivate it. 

The coil you have is given at 20 ohms. The LED resistor is 560 ohms. Together, they are creating a voltage divider that doesn't swing low enough to turn off the coil.

I tried dropping that down to 56 ohms and the relay started switching.

Personally, I'd put that LED indicator somewhere else and use an appropriate resistor on the transistor emitters. Either that or use a relay with more series resistance for the coil.

You will also want to include a fly back diode across the coil to avoid frying your transistors with back emf spikes.

Hope this helps!

nocandopdx, 2024-12-28:
Thank you, that cleared it up : )

voltsimapp, 2025-02-18:
Thank you @cjmventer. Closing this as the question has been answered.

closed 2025-02-18 by voltsimapp