VoltSim/VoltSim #97: Is it possible to add a 555timer example into the app

ElectroBoy10, 2025-02-07:
I have made a 555 timer flasher unit if possible can it be added to examples it can also be edited to look or function better

https://www.voltsimulator.com/dl?id=PCU50HVhDj2zzlbAC6tW

voltsimapp, 2025-02-08:
Hello, Sure we can add it. 
Can you please give a short description and possibly add some text in the circuit explaining what the switches and potentiometer does?

ElectroBoy10, 2025-02-09:
I have made improvements and added txt can u rate it and if it's ok to add it

https://www.voltsimulator.com/dl?id=ufGMeXO4n975qvXalCm8

voltsimapp, 2025-02-10:
Thank you for updating the circuit. We will include this in the examples in our next update. 

We will add more formatting options to text, thank you for the feedback!

ElectroBoy10, 2025-02-10:
Thank you should I close the comments

closed 2025-02-10 by ElectroBoy10