VoltSim/VoltSim #99: Enlarge display clear acquisition

DadoTC, 2025-02-09:
When I dbl tap the plot display to enlarge It the graph disappears , It would be nice ti keep It. I usually start the simulation with a small display for bigger work space and then i enlarge the plot to analyse the acquisition 

Thanks

voltsimapp, 2025-02-10:
Does the graph stay blank after enlarging? can you please share a screen recording or screenshot?

DadoTC, 2025-02-10:
No, after enlarging the graph it restarts working fine.
I will add a clarification, the graph does not disappear only when I enlarge the display but also when I reduce it.
Each time you switch states the graph is lost.

Here the video:

https://github.com/user-attachments/assets/de90a507-5584-43b0-95c2-c9e12a2484f3

voltsimapp, 2025-05-04:
Released in 0.3.7

closed 2025-05-04 by voltsimapp