VoltSim/VoltSim #100: 💌  Inviting Beta testers

voltsimapp, 2025-02-12:
We're gearing up for a major release of VoltSim and are looking for beta testers. Would you be interested in joining?

As a token of appreciation, you'll receive a promo code for the Extensions. Plus, you'll get early access to the latest features and have the opportunity to directly influence VoltSim's development.

If you are interested in joining the Beta test please reach out to voltsim.app@gmail.com for further instructions. 

cc @et-ATK @cjmventer @ElectroBoy10 @UltimateSheep @Explrr @derei @Unexian

Thanks,
VoltSim Team

ElectroBoy10, 2025-02-12:
😁 cool I sent a email

DadoTC, 2025-02-12:
I already joined in the play store 👍🏻

et-ATK, 2025-02-12:
Oh absolutely, I'd love to join!

Unexian, 2025-02-12:
I've also sent a response!

voltsimapp, 2025-02-12:
Hi @DadoTCm This would be a closed beta which is different from the open beta on play store.

ElectroBoy10, 2025-02-12:
> Hi @DadoTCm This would be a closed beta which is different from the open beta on play store.

Ohh I thout that it's also a playstore beta.
I never heard about a closed beta, that's cool.
😁

DadoTC, 2025-02-12:
Ok let me know the details please

Inviato da Outlook per Android<https://aka.ms/AAb9ysg>

ElectroBoy10, 2025-02-16:
@et-ATK @cjmventer @ElectroBoy10 @UltimateSheep @Explrr @derei @Unexian 

Please join the discord server to have direct contact with the dev. https://discord.gg/nFYmtb9T
And get access to the insider beta

Unexian, 2025-02-16:
> @et-ATK @cjmventer @ElectroBoy10 @UltimateSheep @Explrr @derei @Unexian 
> 
> Please join the discord server to have direct contact with the dev. https://discord.gg/nFYmtb9T
> And get access to the insider beta
Already in, just waiting

closed 2025-03-13 by voltsimapp