VoltSim/VoltSim #104: OpAmp power supply pins inverted?

DadoTC, 2025-02-16:
It appears that the power supply pins of the op-amp in VoltSim are reversed from how they usually connect.

Usually the top pin is connected to the positive supply and the bottom pin to the negative supply, like this image:

[image]

From the tests done, I understand instead that in VoltSim they are reversed, is there a specific reason or is it a bug?

voltsimapp, 2025-02-17:
There is not specific reason for this. But now difficult to change this without breaking existing circuits. We will add an option to allow the swapping of rail pins.

voltsimapp, 2025-05-04:
Released in 0.3.7

closed 2025-05-04 by voltsimapp