VoltSim/VoltSim #106: wiring enhancements request (auto-node, auto-component-wiring, manual route wiring)

DadoTC, 2025-02-22:
Hi,
I have 3 wiring suggestions:

1) auto node

Currently, if I want to connect a component to the middle of an existing wire, I need to select the wire, click "split" which creates a node (a junction), move the node/junction to the desired position, and then connect the wire to it.

To make schematic creation quicker and more user-friendly, I'd like to be able to connect directly to an existing wire at any point and have a junction automatically created in that position, as happens in many other electrical design software.

https://github.com/user-attachments/assets/94100adc-9d48-432e-ac16-bf80c33073b8

2) auto component wiring

Additionally, another useful feature would be the ability to drag a component (e.g., a resistor) onto an existing wire—where both terminals are on the same wire—and have the wire automatically split, with the resistor terminals connected accordingly.

https://github.com/user-attachments/assets/325408e2-935a-411b-a06d-f0c58fc6f6fd

3) manual route wiring
Another improvement I'd like to see is the ability to manually define wire routing segment by segment, rather than the current auto-wiring method (clicking the source node and the target node), which often requires adjustments to achieve the desired result.

Thanks!

voltsimapp, 2025-03-02:
A variant of 2. is being tested in beta. and would be released in the next update.  It will create junctions but you would still need to select the middle wire and delete it.

DadoTC, 2025-03-03:
> A variant of 2. is being tested in beta. and would be released in the next update. It will create junctions but you would still need to select the middle wire and delete it.

If I understand well the way it will works, I think this will be a little bit confusing (and will appears as a bug to many users) and can lead to malfunctioning, I mean, if one doesn't notice that the wire is still there or forget to delete it, he will se the circuit with the component in place, but the component will be short-circuit so bypassed/not working .
imho it would be clearer with a full solution or nothing (keeping it as it's now).

thanks