VoltSim/VoltSim #107: Request: plot free vertical zoom

DadoTC, 2025-02-25:
Hi
now in the plot I can only change the horizontal zoom, swiping the screen left and right . I would like to be able to do the same on the vertical axis.
At the moment the plot automatically zooms on the vertical axis based on min and max voltage values, this Is not always what i want, i.e. when I'm looking to a rectified sin voltage with a ripple , the ripple got zoomed but I would like to be able too look to the whole sin graph with the small ripple on top of It. Is there a way to do It now ? I don't find It

Thanks

voltsimapp, 2025-03-02:
Hello, currently we only have auto-scale for the vertical axis. We have plans for adding more controls to give fine-grained control for the oscilloscope scale.

DadoTC, 2025-03-03:
> Hello, currently we only have auto-scale for the vertical axis. We have plans for adding more controls to give fine-grained control for the oscilloscope scale.

I can't wait for them to arrive. 
I actually wanted to ask for a whole set of advanced features on the oscilloscope as found on a physical device (triggering options, dc/ac coupling, horizontal position, vertical position, performing math operations between channels, min, max and many others) but I didn't want to overdo it and only asked for vertical scaling, maybe in the future I will open a new post for advanced features 
thanks !